ralmn/OctoPrint-Ikea-tradfri #8: Logic around error handling on setting page inconsistent.

pedrocadiz13, 2020-05-25:
I've configured the Tradfri  gateway and security code. Seem to have a connection (As the "Connection error with  your Ikea Gateway. Save for retry" message is not showing).

I modified the page locally to see the dropdown for the printer outlets, it's empty (even although I do have  an Ikea Tradfri Outlet)

However  the "Successfully connected with your Ikea Gateway but no outlet found. Save for rescan" message is not being shown. This  conflicts with the fact that the dropdown is empty and
suggests that an outlet was found but not added to the drop down.

WIth the page as is, this means that only the gateway fields and the set-up instructions are visible, which makes it not clear whether or not the connection has succeeded.

ralmn, 2020-05-25:
Do you have any errors messages in `~/.octoprint/logs/octoprint.log` ?

pedrocadiz13, 2020-05-25:
Didn't see anything in the released version, but downloaded the version you mentioned in issue #9  have attached extract from log.

[log.txt](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/4679058/log.txt)

The new wizard looks like an improvement, but has introduced a serious bug; if there are errors you can't close the dialogue, even in a new browser and after a restart, dialog opens up.

ralmn, 2020-05-26:
I fixed the bug you had. Can you reinstall from the master and retry ?

pedrocadiz13, 2020-05-26:
Downloaded from master, but not sure I'm seeing the right version, as I'm no longer seen the wizard. Still not progressing past first step. Have attached new log.

[log.txt](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/4684067/log.txt)

Regards

ralmn, 2020-05-27:
This is strange... Do you have put the right gateway ip and security code ?

pedrocadiz13, 2020-05-27:
I've definitely got  the correct ip and security  code, tested with:

  /usr/local/bin/coap-client/coap-client -m post -u "Client_identity" -k "<security code>" -e "{\"9090\":\"Pi\"}" "coaps://1<gateway ip>:5684/15011/9063"

As per instructions here: https://github.com/glenndehaan/ikea-tradfri-coap-docs#authenticate

I've noticed that this only  returns the shared key the first time it is called for a particular user. I think the problem may be that the line:

      return data['9091']

in __init__.py is not able to extract  the psk as the above has been called previously.

I think you might need to save  the psk on first save and a new one when it expires.

Hope this  is useful.

ralmn, 2020-05-27:
If you try with command line what is your full output ?

ralmn, 2020-05-27:
I had more logs on 5eba01b for try to see where is your problem

pedrocadiz13, 2020-05-27:
Above command outputs:

decrypt_verify(): found 24 bytes cleartext
decrypt_verify(): found 51 bytes cleartext
{"9091":"<key  removed>","9029":"1.10.0036"}

It's getting a key, but because the json is preceded by a couple of lines, json.loads fails.

Changing the code from line 52 to this:

            p = capture_stdout(api)
            result = p.stdout.text
            result  = result.replace("\n","")
            result = re.sub(r'^.*?{', '{', result)
            try:

                data = json.loads(result)
                return data['9091']

Now extracts the key correctly, so gets a bit further, but there's a similar issue in run_gateway_get_request

ralmn, 2020-05-27:
You have debug output on coap. 
Try to reconfigure et rebuild with : 
```
./configure --disable-documentation --disable-shared --without-debug CFLAGS="-D COAP_DEBUG_FD=stderr"
make
sudo make install
```

pedrocadiz13, 2020-05-27:
It took a couple of attempts to get coap to install without debug, had to manually delete and start from scratch.

Working now though.

Thanks for the help.

closed 2020-05-28 by ralmn