ralmn/OctoPrint-Ikea-tradfri #14: System comands of the plugin

ScharV, 2020-05-26:
Is there a way to use the Plugin with System comands trough command line? Would be nice to have the system comands for on and off.
Thanks for this great plugin 🤙🏼

ralmn, 2020-05-26:
I will look how do that :) 
But as workaround you can use the api with curl : `curl -v -X POST https://<octoprint url>api/plugin/ikea_tradfri -H "Content-Type: application/json" -d '{"command": "turnOn"}'` and `curl -v -X POST https://<octoprint url>api/plugin/ikea_tradfri -H "Content-Type: application/json" -d '{"command": "turnOff"}'`

ralmn, 2020-05-26:
Ok i can do that for 1.3 probably (1.2 today or tomorrow)

ScharV, 2020-05-26:
Yeah that looks great😬 will test it later🤙🏼 Thanks

ScharV, 2020-05-26:
I use Octopod for iOS mostly so I can use this comands  with System comand editor to switch the outlet through the app🤙🏼

ralmn, 2020-05-31:
I have add system commands for turn on `octoprint plugins ikea_tradfri:turnon` and turn off `octoprint plugins ikea_tradfri:turnoff` with #18

closed 2020-05-31 by ralmn

ScharV, 2020-05-31:
@ralmn Thanks😍

pierrebv, 2021-11-26:
Hi. I'm sorry, my question is probably stupid, but… I just installed _Custom Control Editor_ and I would like to add a button to turn my outlet on from OctoPod. But I basicaly… don't kown how to use _Custom Control Editor_, apparently. 

I tried to put `octoprint plugins ikea_tradfri:turnon`, or just `ikea_tradfri:turnon`, as good as `curl -v -X POST https://<octoprint url>api/plugin/ikea_tradfri -H "Content-Type: application/json" -d '{"command": "turnOn"}'`, as a _Command_ or a _Script Command_ on _Custom Control Editor_, but each time, the brand new button is not active on Octprint and OctoPod say it's not working because the printer is offline (surprise!).

[image]

I understand I'm writing something to be sent to the printer, while I want to send this command to Octoprint instead. 

I would be ready to believe that's not possible, but this topic and request prove it's possible (my researches drove me originally to this topic, then I downloaded _CCE_ because I'm not confident enough to edit _config.yaml_ by myself). I just don't understand how / I'm probably just missing a little thing. If someone could help me…

PR #18 (2020-05-31): Add system commands
closes #14