ralmn/OctoPrint-Ikea-tradfri #27: Shutdown is not planned but shutdown occurs after print is finished.

poulanker, 2020-11-21:
No shutdown has been planned (I have never used the shutdown feature ). 
After upgrading to V1.4.1 it turns off the outlet about 25 seconds after the print is finished. This was not an issue in V1.4.0

I have tried enabling the feature just to see if that made any changes to the behaviour. It shuts down as expected but the issue persists when finishing another print.

poulanker, 2020-11-22:
I have now reverted to V1.3.2, cleaned the data, filled out gateway info and disabled shutdown and then did a print. 
Now it keeps the outlet turned on as expected.

Apparently it has to do with migrating from 1.3.2 to 1.4.1. Perhaps some variables are not initialized ?

ralmn, 2020-11-22:
Yes i think it was on migration, i set at `True` the value `on_done` when it wasn't set. I will change this... In 1.4.1 if you uncheck the "on done" case it's good ?

poulanker, 2020-11-23:
Yes. I think that's why it shutdown, because I didn't go into settings after migrating from V1.3.2 to V1.4.1. It also displayed "No shutdown planned" so I thought it was ok, but it did shutdown anyway.

Now, that I have migrated again and unchecked the "on done", it doesn't shutdown unexpectedly.
I still think it's a bug though, because the migration fails to identify the existing settings correctly.

I do think that your plugin is great though. I used to turn the printer on and off from another room using voice commands to Google home.  It's so much nicer using the plugin. 
I will also begin to use the shutdown feature. I often leave the printer on longer than necessary when a print has finished.

ralmn, 2020-11-28:
V1.4.2 will fix this now

closed 2020-11-28 by ralmn

poulanker, 2020-11-28:
great, thanks :-)