ralmn/OctoPrint-Ikea-tradfri #32: Can't update

homan85, 2021-01-12:
Trying to update from 1.4.4 to 1.5.0 but receive this error:
Updating, please wait.
+++++++++++++++++++++++++++++++++++++++++
Now updating Ikea Tradfri Plugin to 1.5.0
+++++++++++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/1.5.0.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/1.5.0.zip
Downloading https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/1.5.0.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Ikea-Tradfri==1.5.0) (1.5.2)
Requirement already satisfied: sarge in /home/pi/oprint/lib/python2.7/site-packages (from OctoPrint-Ikea-Tradfri==1.5.0) (0.1.5.post0)
Collecting aiocoap[all]@ https://github.com/chrysn/aiocoap/archive/master.zip
Downloading https://github.com/chrysn/aiocoap/archive/master.zip
ERROR: Package 'aiocoap' requires a different Python: 2.7.16 not in '>=3.6.9'
The update did not finish successfully. Please consult <code>octoprint.log</code> and <code>plugin_softwareupdate_console.log</code> for details.

Result of python commands:
python --version
Python 2.7.16

python3 --version
Python 3.7.3

For fun I tried to uninstall Python2.7, but then OctoPi wouldn't start.

ralmn, 2021-01-12:
Hello, you need to upgrade your octoprint to python3  : https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

cp2004, 2021-01-12:
@ralmn to avoid some of these issues, explicitly set `python_requires` in your setup parameters:
```diff
- additional_setup_parameters = {}
+ additional_setup_parameters = {"python_requires": ">3.6,<4"}
```
This means that the plugin will refuse to install with Python 2 at the earliest opportunity, without *looking* like it will install. It is not going to be easy but this might help. Some people still will not read your release notes, as hard as you try.

The alternative would be to add some logic to the setup.py file that would print something like "Please upgrade to Python 3 to use this plugin" if `sys.version_info.major` is not 3.

homan85, 2021-01-12:
> Hello, you need to upgrade your octoprint to python3 : https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

Hi. I already have Python3 installed on the Pi. Since Python2 cannot be removed, it will just look and find the first best Python, and it happens to be Python2. And it stops there and refuse to install since Python3 was not there.
If it had looked a little more it would find Python3 aswell.

So the upgrade/install of Python doesn't solve anything. Sorry.

I think what CP2004 said is the right way to go.

cp2004, 2021-01-12:
@homan85 did you read the link that they posted? You can't just have Python 3 installed, OctoPrint has to use it. So please follow the instructions to upgrade your instance.

ralmn, 2021-01-12:
Thanks for your suggestion @cp2004.  I just released 1.5.2 with the modification

closed 2021-01-14 by ralmn