ralmn/OctoPrint-Ikea-tradfri #33: Update to 1.5.2 not recognized

Thidsa, 2021-01-14:
It installs without error, but after octoprint restart it has not updated.
Tell me what more you need
Bjørn

ralmn, 2021-01-14:
What is the version of python used by octoprint? 
For information: version 1.5.x of the plugin is no longer compactible with python2.7.

Thidsa, 2021-01-14:
Python 3.7.3
Current version of your plugin is 1.5.1 here

ralmn, 2021-01-14:
Can you join the `octoprint.log`, `plugin_pluginmanager_console.log` and `plugin_softwareupdate_console.log` ?

Thidsa, 2021-01-14:
plugin_pluginmanager_console.log was empty here......

[plugin_softwareupdate_console.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/5817323/plugin_softwareupdate_console.log)
[octoprint.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/5817324/octoprint.log)

ralmn, 2021-01-15:
Ok... I found what happened... I have forgot to change a version in setup.py... The 1.5.3 fix that

closed 2021-01-15 by ralmn

Thidsa, 2021-01-15:
yes works now, thanks