ralmn/OctoPrint-Ikea-tradfri #37: Issue with Python version - Probleme avec la version du python

Dedcap, 2021-02-05:
Hello,

When I try to install it from the zip, it throw me due to a bad version of python
If I try to download the file my self, it don't throws me anymore for the python but after the restar of Octo, it crashes and oblige me to restart in safe mode and uninstall the plugin

Can you advise me how to proceed?

Thanks

Dedcap, 2021-02-05:
Hi,

I have updated octo to have it with python 3
Now the plugin is installed and I can click on the icon to switch off and on the Ikea plug

Is it possible to have octo switching off the plug at the end of a print?
I try to configure this with your plugin but unfortunately it look like to do know works

ralmn, 2021-02-14:
I close this issue, your problem is solved

closed 2021-02-14 by ralmn