ralmn/OctoPrint-Ikea-tradfri #39: Wizard unable to finish on one of my OctoPis

Larswa, 2021-03-06:
Hi, I have two Octopis, both running Octoprint 1.5.3   One is a on new 8GB RP4 and the one that I have a problem with is an RP3B+ with just a gig of RAM.

On the RP4, the Ikea plugin is working fine. But on the other I have a UI problem.

I install the plugin, and on restart of the server I am forced into the wizard. I give it my IP, and the API key, and it connects to my Traadloes hub just fine. The dropdown is correctly populated with what the Hub can see. I select an item from the list, and click finish.  Nothing happens.  I can see this in the browser console.

``` javascript
packed_core.js?22357642:431 Uncaught TypeError: Cannot read property 'done' of undefined
    at WizardViewModel.self.finishWizard (packed_core.js?22357642:431)
    at Object.onFinish (packed_core.js?22357642:430)
    at HTMLButtonElement.finish (packed_libs.js?ab739628:1209)
    at HTMLButtonElement.dispatch (packed_libs.js?ab739628:1)
    at HTMLButtonElement.v.handle (packed_libs.js?ab739628:1)
```
Here is a screenshot of the wizard 
https://photos.app.goo.gl/YY1eJSqDTqFLq12HA

I have recreated this in Brave, Google Chrome, Firefox and edge. Same behavior in all browsers.  And I have no idea why I am only seeing this on one of my octopis.

I have to ssh into my octopi instance and uninstall it manually as I can't get past the wizard, every time I load the page.
`~/oprint/bin/pip uninstall OctoPrint-Ikea-Tradfri`
`sudo systemctl restart octoprint`

Is there something else I need to do to cleanup the old instance and get a fresh one installed?

ralmn, 2021-03-06:
What is your octoprint's python version ? 
Can you send me the `octoprint.log` ? (`~/.octoprint/logs/octoprint.log`)

Larswa, 2021-03-06:
Python version is 3.7.3 

Here is a link to the octoprint.log
https://gofile.me/6y5OS/V0OJzAUOY

I believe that when I installed it the first time, I fumbled around with the API keys ... that's the only difference I can tell between the instance I have working on the other OctoPi.

Larswa, 2021-03-06:
Also ... a really cool plugin. Thanks for that. @ralmn

ralmn, 2021-03-06:
Can you try to uninstall plugin, remove all ikea_tradfri reference in config.yaml, and reinstall plugin ? 

Your javascript error is an error of octoprint...

Larswa, 2021-03-06:
I did that. Still getting the same error. Should I just reflash the Octopi card on the Raspberry Pi as this is an Octoprint error?

ralmn, 2021-03-06:
Maybe... I dont think it's a plugin error...

Larswa, 2021-03-06:
Me neither ... I also has it working fine on another instance. Just wanted to make sure that I had done all I could, and removed every trace of the plugin before I reinstalled.  Lets close this issue.

andreer, 2021-03-24:
This happened to me too when setting up a fresh install of Octoprint 0.18.0 - I got stuck on the last step of the wizard and couldn't continue - pressing the "Finish" button did nothing.

This is likely the same as #39, but I am able to consistently reproduce it by installing ikea_tradfri:1.7.0, printjobhistory:1.10.0, and spoolmanager:1.3.3. So there's some kind of conflict that happens with one (or both) of these plugins.  @Larswa, perhaps you installed those too?

If I install ikea_tradfri first, and complete the setup wizard before installing those plugins, it works.

Larswa, 2021-03-24:
Hi @andreer   That could very well be it. I will test later today when my current print is finished, and update this thread. I traced my original error to a dependency of ikea_tradfri that failed, but never got around to follow that trail to the end. So it definitely smelled like a dependency collision of some sorts.

Larswa, 2021-03-24:
Brilliant! It worked. Removed Spoolmanager and install and configuration of Ikea Tradfri worked.  After that I reinstalled Spoolmanager, and everything still seems to be working.  Thanks @andreer

ralmn, 2021-03-24:
Can i have yours `octoprint.log` and `plugin_pluginmanager_console.log` @Larswa @andreer  ? I would like see if i can fix it....

andreer, 2021-03-26:
To see if I could reproduce it from scratch, I actually flashed the Octopi 0.18.0 image to a fresh SD card using balenaEtcher, installed the three plugins mentioned above from the , and restarted. And I am indeed now stuck in the setup wizard in the same way. Here are the logs:

[octoprint.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/6211783/octoprint.log)
[plugin_pluginmanager_console.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/6211784/plugin_pluginmanager_console.log)

And in my browser console I see the same stack trace appear each time I click "Finish":
```
Uncaught TypeError: self.settingsViewModel.saveData() is undefined
    finishWizard http://octopi.local/static/webassets/packed_core.js?b8ae0794:431
    onFinish http://octopi.local/static/webassets/packed_core.js?b8ae0794:430
    finish http://octopi.local/static/webassets/packed_libs.js?ab739628:1209
    dispatch http://octopi.local/static/webassets/packed_libs.js?ab739628:1
    handle http://octopi.local/static/webassets/packed_libs.js?ab739628:1```

I'd be happy to give you web/ssh access to the pi to debug, just send me an email (or twitter DM) and I'll give you the details.

ralmn, 2021-03-28:
The probable cause is  https://github.com/OllisGit/OctoPrint-SpoolManager/blob/master/octoprint_SpoolManager/static/js/SpoolManager.js#L295

ralmn, 2021-03-28:
I think this PR will fix the issue : https://github.com/OllisGit/OctoPrint-SpoolManager/pull/143

andreer, 2021-03-31:
Thanks for the fix, and in another plugin at that! Hopefully it'll be merged soon, that may be causing trouble in other cases too. I saw you fixed the (unrelated) MQTT exception in the latest version also 👍

ralmn, 2021-04-17:
OllisGit/OctoPrint-SpoolManager#143 is merged and [v1.3.4](https://github.com/OllisGit/OctoPrint-SpoolManager/releases/tag/1.3.4) is released

closed 2021-04-17 by ralmn