ralmn/OctoPrint-Ikea-tradfri #40: Update / uninstall via octoprint GUI doesn't work.

pedrocadiz13, 2021-03-07:
Since V1.5.4 update to newer version appears to work, but   v1.5.4 is still displayed in Plugin Manager. Attempting to uninstall to do a clean install fails uninstall command contains unknown option:

/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install --yes -ctoPrint-Ikea-Tradfri

Should be 

/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install --yes octoPrint-Ikea-Tradfri

Even after  running this from the terminal and installing with:

/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/1.6.0.zip

v1.5.4 is still displayed in Plugin Manager, and I'm prompted to update after every log in.

ralmn, 2021-03-07:
Can you attach octoprint logs (`octoprint.log`) ?

pedrocadiz13, 2021-03-07:
[octoprint.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/6097879/octoprint.log)

Log attached  as  requested. Hope this helps.

ralmn, 2021-03-07:
I really don't see where this can come from... 
Can I have your `config.yaml`, `plugin_pluginmanager_console.log` and `plugin_softwareupdate_console.log`  ?

pedrocadiz13, 2021-03-07:
[plugin_pluginmanager_console.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/6098157/plugin_pluginmanager_console.log)
[plugin_softwareupdate_console.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/6098158/plugin_softwareupdate_console.log)

ralmn, 2021-03-07:
he says that he managed to install the 1.6.0... Looks like you have a version 1.5.4 and 1.6.0 installed at the same time...

Can I have an ouput of `ls -al /home/pi/oprint/lib/python3.7/site-packages` and `/home/pi/oprint/bin/python3 -m pip list` ?

pedrocadiz13, 2021-03-07:
., .., aiocoap, aiocoap-0.4.1.post0.dist-info,
Arc_Welder-1.0.0+u.bb71e8f.dist-info, babel, Babel-2.9.0.dist-info, backports,
backports.csv-1.0.7.dist-info, backports.csv-1.0.7-py3.7-nspkg.pth, bitlist,
bitlist-0.3.1.dist-info, blinker, blinker-1.4.dist-info, cachelib,
cachelib-0.1.1.dist-info, cbor2, cbor2-5.2.0.dist-info,
_cbor2.cpython-37m-arm-linux-gnueabihf.so, certifi, certifi-2020.12.5.dist-info,
cffi, cffi-1.14.5.dist-info, _cffi_backend.cpython-37m-arm-linux-gnueabihf.so,
Change_Filament-0.3.2.dist-info, chardet, chardet-4.0.0.dist-info, click,
click-7.1.2.dist-info, complicated, complicated-1.1.0.dist-info, cryptography,
cryptography-3.4.5.dist-info, '~ctoprint_ikea_tradfri',
'~ctoPrint_Ikea_Tradfri-1.5.4.dist-info', Cython, Cython-0.29.21.dist-info,
cython.py, DisplayLayerProgress-1.25.4.dist-info, _distutils_hack,
distutils-precedence.pth, DTLSSocket, DTLSSocket-0.1.12.dist-info,
easy_install.py, emoji, emoji-0.6.0.dist-info, examples, fe25519,
fe25519-0.2.0.dist-info, feedparser, feedparser-6.0.2.dist-info,
filelock-3.0.12.dist-info, filelock.py, filetype, filetype-1.0.7.dist-info,
flask, Flask-1.1.2.dist-info, Flask_Assets-2.0.dist-info, flask_assets.py,
flask_babel, Flask_Babel-1.0.0.dist-info, flask_login,
Flask_Login-0.5.0.dist-info, fountains, fountains-0.2.1.dist-info, frozendict,
frozendict-1.2.dist-info, future, future-0.18.2.dist-info, ge25519,
ge25519-0.2.0.dist-info, gntp, gntp-1.0.3.dist-info, idna, idna-2.10.dist-info,
ifaddr, ifaddr-0.1.7.dist-info, itsdangerous, itsdangerous-1.1.0.dist-info,
jinja2, Jinja2-2.11.2.dist-info, libfuturize, libpasteurize,
LinkHeader-0.4.3.dist-info, link_header.py, markdown, Markdown-3.1.1.dist-info,
markupsafe, MarkupSafe-1.1.1.dist-info, netaddr, netaddr-0.8.0.dist-info,
netifaces-0.10.9.dist-info, netifaces.cpython-37m-arm-linux-gnueabihf.so,
octoprint, OctoPrint-1.5.3.dist-info, octoprint_arc_welder,
octoprint_arc_welder_setuptools, octoprint_autoscroll,
OctoPrint_Autoscroll-0.0.3.dist-info, octoprint_BetterHeaterTimeout,
OctoPrint_BetterHeaterTimeout-1.3.0.dist-info, octoprint_Change_Filament,
octoprint_client, octoprint_complicated, OctoPrint_Complicated-1.1.3.dist-info,
octoprint_costestimation, OctoPrint_CostEstimation-3.2.0.dist-info,
octoprint_dashboard, OctoPrint_Dashboard-1.17.0.dist-info,
octoprint_DisplayLayerProgress, octoprint_displayz,
OctoPrint_DisplayZ-0.2.0.dist-info, octoprint_dragon_order,
OctoPrint_DragonOrder-0.1.7.dist-info, octoprint_excluderegion,
OctoPrint_ExcludeRegionPlugin-0.3.0.dist-info, octoprint_filamentmanager,
OctoPrint_FilamentManager-1.7.2.dist-info, octoprint_file_check,
OctoPrint_FileCheck-2021.2.23.dist-info, octoprint_firmware_check,
OctoPrint_FirmwareCheck-2021.2.4.dist-info, octoprint_fullscreen,
OctoPrint_Fullscreen-0.0.6.dist-info, octoprint_growl,
OctoPrint_Growl-0.1.1.dist-info, octoprint_ikea_tradfri,
OctoPrint_Ikea_Tradfri-1.6.0.dist-info, octoprint_multi_colors,
OctoPrint_MultiColors-1.0.17.dist-info, octoprint_navbartemp,
OctoPrint_NavbarTemp-0.14.dist-info, octoprint_octopod,
OctoPrint_OctoPod-0.3.0.dist-info, octoprint_preheat, octoprint_PrintTimeGenius,
OctoPrint_PrintTimeGenius-2.2.7.dist-info, octoprint_progress_title,
OctoPrint_Progress_Title-0.1.1.dist-info, octoprint_psucontrol,
OctoPrint_PSUControl-0.1.11.dist-info, octoprint_setuptools,
octoprint_simpleemergencystop, OctoPrint_SimpleEmergencyStop-1.0.4.dist-info,
octoprint_systemcommandeditor, OctoPrint_SystemCommandEditor-0.3.4.dist-info,
octoprint_taborder, OctoPrint_TabOrder-0.5.12.dist-info, octoprint_themeify,
octoprint_translatemodel, OctoPrint_TranslateModel-0.1.1.dist-info,
octoprint_ultimakerformatpackage,
OctoPrint_UltimakerFormatPackage-0.2.2.dist-info, octoprint_webcamtab,
OctoPrint_WebcamTab-0.2.0.dist-info, octoprint_widescreen,
OctoPrint_WideScreen-0.1.4.dist-info, octoprint_wifistatus,
OctoPrint_WiFiStatus-1.5.1.dist-info, parts, parts-1.0.2.dist-info, past,
pathtools, pathtools-0.1.2.dist-info, peewee-3.14.1.dist-info, peewee.py, PIL,
Pillow-8.1.0.dist-info, pip, pip-21.0.1.dist-info, pkginfo,
pkginfo-1.6.1.dist-info, pkg_resources, pkg_resources-0.0.0.dist-info,
playhouse, Preheat-0.6.0.dist-info, psutil, psutil-5.8.0.dist-info, pwiz.py,
PyArcWelder.cpython-37m-arm-linux-gnueabihf.so, pyasn1, pyasn1-0.4.8.dist-info,
__pycache__, pycparser, pycparser-2.20.dist-info, pygments,
Pygments-2.7.4.dist-info, pylru-1.2.0.dist-info, pylru.py,
pyserial-3.5.dist-info, pytz, pytz-2020.5.dist-info, pyximport,
PyYAML-5.3.1.dist-info, regex, regex-2020.11.13.dist-info, requests,
requests-2.25.1.dist-info, RPi, RPi.GPIO-0.7.0.dist-info, rsa,
rsa-4.0.dist-info, sarge, sarge-0.1.5.post0.dist-info, semantic_version,
semantic_version-2.8.5.dist-info, sentry_sdk, sentry_sdk-0.19.5.dist-info,
serial, setuptools, setuptools-51.1.2.dist-info, setuptools_rust,
setuptools_rust-0.11.6.dist-info, sgmllib3k-1.0.0.dist-info, sgmllib.py,
six-1.15.0.dist-info, six.py, sqlalchemy, SQLAlchemy-1.1.18.dist-info,
termcolor-1.1.0.dist-info, termcolor.py, Themeify-1.2.2.dist-info, toml,
toml-0.10.2.dist-info, tornado, tornado-5.1.1.dist-info,
translate.cpython-37m-arm-linux-gnueabihf.so, unidecode,
Unidecode-0.04.21.dist-info, uritools, uritools-2.1.1.dist-info, urllib3,
urllib3-1.26.2.dist-info, watchdog, watchdog-0.10.4.dist-info, webassets,
webassets-2.0.dist-info, websocket, websocket_client-0.57.0.dist-info,
websockets, websockets-8.1.dist-info, werkzeug, Werkzeug-1.0.1.dist-info, wheel,
wheel-0.36.2.dist-info, wrapt, wrapt-1.12.1.dist-info, yaml, zeroconf,
zeroconf-0.24.5.dist-info, zeroconf.py

ralmn, 2021-03-07:
I don't know why you have an `~ctoprint_ikea_tradfri`... 
Try to uninstall `/home/pi/oprint/bin/python3 -m pip uninstall --yes '~ctoprint_ikea_tradfri'`

ralmn, 2021-03-07:
And after restart your octoprint instance

pedrocadiz13, 2021-03-07:
Running  the ab one command results in:

ERROR: Invalid requirement: '~ctoprint_ikea_tradfri'

ralmn, 2021-03-08:
Can you try to delete folder `~ctoprint_ikea_tradfri` and `~ctoPrint_Ikea_Tradfri-1.5.4.dist-info` ? and restart your octoprint ?

pedrocadiz13, 2021-03-08:
That  seems to have done the  trick.

Many thanks.

closed 2021-03-08 by pedrocadiz13