ralmn/OctoPrint-Ikea-tradfri #41: Outlet icon is always red 1.7.0

Thidsa, 2021-03-25:
Icon doesnt turn green anymore when outlet is on. Not sure since when

nestorix1343, 2021-03-26:
> Icon doesnt turn green anymore when outlet is on. Not sure since when

My feeling is since 1.7.0. 

This has happened before I think and it got fixed quite fast.

ralmn, 2021-03-26:
On one of my printer, it's work... But I will see in more detail why it's happend... Can I have your `octoprint.log` ?

ralmn, 2021-03-26:
I just had the problem... It seems that it is the ikea gateway that did not respond in my case... A periodic check should help... even if it's not the best...

Thidsa, 2021-03-27:
Respond how?
Here it works like it should, atleast the controlling of the plug. Can try to restart the gateway
Still need a log?

ralmn, 2021-03-27:
My gateway timeout. 
Yes if you can send me logs, for see if it's the same error

Thidsa, 2021-03-27:
[octoprint.log](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/6216554/octoprint.log)

ralmn, 2021-03-28:
Ok there is a real problem when the MQTT plugin is not enabled. v1.7.1 will fix this :)

ralmn, 2021-03-28:
https://github.com/ralmn/OctoPrint-Ikea-tradfri/releases/tag/1.7.1

closed 2021-03-28 by ralmn