ralmn/OctoPrint-Ikea-tradfri #48: Chrome messing up security key due to password autofill

pellz0r, 2021-05-17:
So I've noticed several times that the plugin stops working, and I just figured out that it's the Google Chrome password manager that automatically replace IP/security code with my Octoprint login credentials instead.

It seem to do this even if I never look at the Trådfri plugin page, so when I change something in Octoprint settings, Chrome change the security code which borks the plugin.

Is it really important to have the security code as a password field? If it's not a password field I assume that Chrome will leave it alone.

ralmn, 2021-05-18:
I have add `autocomplete="off"` on input and form. So can you confirm it's fixing well with update from develop branch ? (Settings, Plugin Manager, Get more, from url https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/refs/heads/develop.zip)

pellz0r, 2021-05-18:
Looking good! Chrome doesn't give credential suggestions when clicking in the fields any more.

commit 607ebe36 (2021-05-18): Add autocomplete=off for deny browser autofill overwrite security code, Fix #48
closed 2021-05-18 by ralmn