ralmn/OctoPrint-Ikea-tradfri #50: octoprint cpu to 100% when gateway not responsive

redPanther, 2021-11-26:
Hi,
in general this plugin works great. But sometimes, when the tradfri gateway is not reachable or not reactive, the tradfri plugin raises octoprint cpu to 100% and more - wihlie not printing. When printing this is extremly anoying, because then gcode transfer is not smooth anymore and the printer begins to "stutter".

Raspi and tradfri gateway are connected via cable (no wireless )

plugin version  1.7.5

elaurijssens, 2022-08-03:
Having the same issue. When the gateway is unresponsive, one of the threads goes to 100% and the following is logged tot /var/log/syslog:

```
Aug  3 18:37:23 prusa3d-01 octoprint[26207]: #033[31m2022-08-03 18:37:23,133 - octoprint.plugins.ikea_tradfri - ERROR - _run_gateway_get_request(): Failed to fetch resource:#033[0m
Aug  3 18:37:23 prusa3d-01 octoprint[26207]: #033[31m2022-08-03 18:37:23,134 - octoprint.plugins.ikea_tradfri - ERROR - Retransmissions exceeded#033[0m
Aug  3 18:37:23 prusa3d-01 octoprint[26207]: #033[31m2022-08-03 18:37:23,139 - octoprint - ERROR - Exception on /plugin/ikea_tradfri/navbar/info [GET]#033[0m
Aug  3 18:37:23 prusa3d-01 octoprint[26207]: Traceback (most recent call last):
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     response = self.full_dispatch_request()
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     rv = self.handle_user_exception(e)
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     rv = self.dispatch_request()
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 612, in navbarInfo
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     data = self.navbarInfoData()
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 398, in navbarInfoData
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     state=self.getStateData()
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 800, in getStateData
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     res[device['id']] = self.getStateDataById(device['id'])
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 820, in getStateDataById
Aug  3 18:37:23 prusa3d-01 octoprint[26207]:     if code in data and len(data[code]) > 0 and "5850" in data[code][0]:
Aug  3 18:37:23 prusa3d-01 octoprint[26207]: TypeError: argument of type 'NoneType' is not iterable
Aug  3 18:38:38 prusa3d-01 octoprint[26207]: #033[31m2022-08-03 18:38:38,553 - octoprint.plugins.ikea_tradfri - ERROR - _run_gateway_get_request(): Failed to fetch resource:#033[0m
Aug  3 18:38:38 prusa3d-01 octoprint[26207]: #033[31m2022-08-03 18:38:38,553 - octoprint.plugins.ikea_tradfri - ERROR - Retransmissions exceeded#033[0m
Aug  3 18:38:38 prusa3d-01 octoprint[26207]: #033[31m2022-08-03 18:38:38,557 - octoprint - ERROR - Exception on /plugin/ikea_tradfri/navbar/info [GET]#033[0m
Aug  3 18:38:38 prusa3d-01 octoprint[26207]: Traceback (most recent call last):
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 2077, in wsgi_app
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     response = self.full_dispatch_request()
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1525, in full_dispatch_request
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     rv = self.handle_user_exception(e)
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1523, in full_dispatch_request
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     rv = self.dispatch_request()
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/flask/app.py", line 1509, in dispatch_request
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 612, in navbarInfo
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     data = self.navbarInfoData()
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 398, in navbarInfoData
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     state=self.getStateData()
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 800, in getStateData
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     res[device['id']] = self.getStateDataById(device['id'])
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:   File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_ikea_tradfri/__init__.py", line 820, in getStateDataById
Aug  3 18:38:38 prusa3d-01 octoprint[26207]:     if code in data and len(data[code]) > 0 and "5850" in data[code][0]:
Aug  3 18:38:38 prusa3d-01 octoprint[26207]: TypeError: argument of type 'NoneType' is not iterable
```

DavyGD, 2024-06-11:
I've got the same issue.

ralmn, 2025-05-07:
I regret to inform you that I am no longer able to maintain this plugin due to time constraints and lack of access to the necessary hardware (Tradfri hub) to address issues effectively. (cf #57)

closed 2025-05-07 by ralmn