ralmn/OctoPrint-Ikea-tradfri #51: Install Error

SimoneDeVittori, 2022-01-02:
ERROR: Failed building wheel for cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

OctoPrint 1.7.2 Python 3.7.3 OctoPi 0.18.0

DerBoeck, 2022-01-05:
Same problem here!

pascals92, 2022-01-15:
also here! :/

p0w3rmast3r, 2022-02-08:
Same Problem here!

ralmn, 2022-06-05:
I canno't reproduce on my rapsberrypi (3B+ and 4B) with OctoPrint 1.8.1, Octopi 0.18.0, Python 3.7.3, PIP 20.3.3

Can you send me the Systeminfo bundle? (Settings -> System Informations -> Download Systeminfo Bundle)

g233hacker, 2022-06-05:
here is the  Systeminfo bundle
[octoprint-systeminfo-20220605170743.zip](https://github.com/ralmn/OctoPrint-Ikea-tradfri/files/8840109/octoprint-systeminfo-20220605170743.zip)

g233hacker, 2022-06-05:
new error now
Failed to build DTLSSocket
ERROR: Could not build wheels for DTLSSocket, which is required to install pyproject.toml-based projects
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

Icegames-dk, 2022-09-23:
Hi

octoprint.version: 1.8.3
printer.firmware: Marlin 2.1.1
env.python.version: 3.7.3
env.plugins.pi_support.model: Raspberry Pi 3 Model B Plus Rev 1.3
env.plugins.pi_support.octopi_version: 0.18.0

I have this error:

Building wheel for cryptography (PEP 517): startedERROR: Command errored out with exit status 1:command: /home/pi/oprint/bin/python3 /home/pi/oprint/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmp8kpfhnmvcwd: /tmp/pip-install-tdo09l27/cryptography_5ce89bd56ca64b3386dd730573461617Complete output (193 lines):running bdist_wheelrunning buildrunning build_pycreating buildcreating build/lib.linux-armv7l-cpython-37creating build/lib.linux-armv7l-cpython-37/cryptographycopying src/cryptography/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptographycopying src/cryptography/fernet.py -> build/lib.linux-armv7l-cpython-37/cryptographycopying src/cryptography/utils.py -> build/lib.linux-armv7l-cpython-37/cryptographycopying src/cryptography/__about__.py -> build/lib.linux-armv7l-cpython-37/cryptographycopying src/cryptography/exceptions.py -> build/lib.linux-armv7l-cpython-37/cryptographycreating build/lib.linux-armv7l-cpython-37/cryptography/hazmatcopying src/cryptography/hazmat/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmatcopying src/cryptography/hazmat/_oid.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmatcreating build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/certificate_transparency.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/name.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/general_name.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/base.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/oid.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/extensions.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509copying src/cryptography/x509/ocsp.py -> build/lib.linux-armv7l-cpython-37/cryptography/x509creating build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/poly1305.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/_asymmetric.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/keywrap.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/_cipheralgorithm.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/hashes.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/constant_time.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/cmac.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/hmac.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/padding.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescopying src/cryptography/hazmat/primitives/_serialization.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitivescreating build/lib.linux-armv7l-cpython-37/cryptography/hazmat/backendscopying src/cryptography/hazmat/backends/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/backendscreating build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindingscopying src/cryptography/hazmat/bindings/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/bindingscreating build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/cipherscopying src/cryptography/hazmat/primitives/ciphers/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/cipherscopying src/cryptography/hazmat/primitives/ciphers/algorithms.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/cipherscopying src/cryptography/hazmat/primitives/ciphers/modes.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/cipherscopying src/cryptography/hazmat/primitives/ciphers/aead.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/cipherscopying src/cryptography/hazmat/primitives/ciphers/base.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/cipherscreating build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/kdfcopying src/cryptography/hazmat/primitives/kdf/__init__.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/kdfcopying src/cryptography/hazmat/primitives/kdf/hkdf.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/kdfcopying src/cryptography/hazmat/primitives/kdf/x963kdf.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/kdfcopying src/cryptography/hazmat/primitives/kdf/concatkdf.py -> build/lib.linux-armv7l-cpython-37/cryptography/hazmat/primitives/kdfcopying src/cryptography/hazmat/primitives/kdf/scrypt.py -> build/lib.linux-armv7l-cpython-37/cryptogr [... cut ...] ain.This package requires Rust >=1.48.0.----------------------------------------ERROR: Failed building wheel for cryptographyERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directlyBuilding wheel for cryptography (PEP 517): finished with status 'error'Successfully built OctoPrint-Ikea-Tradfri aiocoapFailed to build cryptographyLooks like the plugin was already installed. Forcing a reinstall.

I hope it helps fixing the problem, the pugin is very use full.

Criticalstone, 2022-10-19:
Hi! I resolved this by installing the following packages manually on the pi 
`sudo apt-get install build-essential libssl-dev libffi-dev python-dev`
Then installing the rust compiler as well as manually installing the cryptography python package 
`/OctoPrint/bin/pip install cryptography`

Thidsa, 2022-11-22:
More or less the same here:

DietPi v8.10.2 : 01:01 - Tue 11/22/22
OctoPrint 1.8.6 Python 3.7.3
rustc 1.65.0 (897e37553 2022-11-02)

After a reinstall of a loong working octoprint, your IKEA plugin dont reinstall with above mentioned problems.
Resolving like @Criticalstone did doesnt work here.

Installing plugin "Ikea Tradfri" from https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/master.zip...
/usr/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpdbsmgtsb/OctoPrint-Ikea-tradfri-master.zip --no-cache-dir --user
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpdbsmgtsb/OctoPrint-Ikea-tradfri-master.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Collecting aiocoap[all]@ https://github.com/chrysn/aiocoap/archive/master.zip
Downloading https://github.com/chrysn/aiocoap/archive/master.zip
\ 344.7 kB 2.2 MB/s 0:00:00
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint-Ikea-Tradfri==1.7.5) (1.8.6)
Requirement already satisfied: sarge in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint-Ikea-Tradfri==1.7.5) (0.1.6)
Collecting LinkHeader
Downloading https://www.piwheels.org/simple/linkheader/LinkHeader-0.4.3-py3-none-any.whl (5.7 kB)
Collecting cbor2
Downloading https://www.piwheels.org/simple/cbor2/cbor2-5.4.3-cp37-cp37m-linux_armv7l.whl (99 kB)
???????????????????????????????????????? 99.7/99.7 kB 1.0 MB/s eta 0:00:00
Collecting cryptography>=2.0
Downloading cryptography-38.0.3.tar.gz (599 kB)
??????????????????????????????????????? 599.9/599.9 kB 8.3 MB/s eta 0:00:00
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Preparing metadata (pyproject.toml): started
Preparing metadata (pyproject.toml): finished with status 'done'
Collecting filelock
Downloading https://www.piwheels.org/simple/filelock/filelock-3.8.0-py3-none-any.whl (10 kB)
Collecting ge25519
Downloading https://www.piwheels.org/simple/ge25519/ge25519-1.3.0-py3-none-any.whl (46 kB)
??????????????????????????????????????? 46.4/46.4 kB 904.5 kB/s eta 0:00:00
Collecting DTLSSocket>=0.1.11a1
Downloading https://www.piwheels.org/simple/dtlssocket/DTLSSocket-0.1.12-cp37-cp37m-linux_armv7l.whl (291 kB)
??????????????????????????????????????? 291.7/291.7 kB 3.3 MB/s eta 0:00:00
Collecting websockets
Downloading https://www.piwheels.org/simple/websockets/websockets-10.4-cp37-cp37m-linux_armv7l.whl (109 kB)
?????????????????????????????????????? 109.4/109.4 kB 21.9 MB/s eta 0:00:00
Collecting termcolor
Downloading https://www.piwheels.org/simple/termcolor/termcolor-2.1.1-py3-none-any.whl (6.2 kB)
Requirement already satisfied: pygments in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from aiocoap[all]@ https://github.com/chrysn/aiocoap/archive/master.zip->OctoPrint-Ikea-Tradfri==1.7.5) (2.13.0)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.3.4 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (1.3.4)
Requirement already satisfied: OctoPrint-PiSupport>=2022.3.28 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (2022.6.13)
Requirement already satisfied: wrapt<1.14,>=1.13.3 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (1.13.3)
Requirement already satisfied: flask<2.2,>=2.1 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (2.1.3)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/local/lib/python3.7/dist-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (3.4)
Requirement already satisfied: pylru<2,>=1.2 in /usr/local/lib/python3.7/dist-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (1.2.0)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (0.33.4)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /usr/local/lib/python3.7/dist-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (0.5.0)
Requirement already satisfied: filetype<2,>=1.0.7 in /usr/local/lib/python3.7/dist-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (1.0.7)
Requirement already satisfied: sentry-sdk<2,>=1.5.7 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (1.11.0)
Requirement already satisfied: feedparser<7,>=6.0.8 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (6.0.10)
Requirement already satisfied: regex in /usr/local/lib/python3.7/dist-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (2020.10.23)
Requirement already satisfied: colorlog<7,>=6 in /mnt/dietpi_userdata/octoprint/.local/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Ikea-Tradfri==1.7.5) (6.7.0)
[... cut ...]
Building wheel for OctoPrint-Ikea-Tradfri (setup.py): finished with status 'done'
Created wheel for OctoPrint-Ikea-Tradfri: filename=OctoPrint_Ikea_Tradfri-1.7.5-py3-none-any.whl size=17530 sha256=ffa9304e727cd52f53895cecbaa157a44f0963c0769249a558f6b69bbc355010
Stored in directory: /tmp/pip-ephem-wheel-cache-jx2ds8sg/wheels/82/51/23/13966e1721c5cecc0474b482273153cd04ac1323180ea2b6d0
Successfully built OctoPrint-Ikea-Tradfri
Installing collected packages: OctoPrint-Ikea-Tradfri
Successfully installed OctoPrint-Ikea-Tradfri-1.7.5
Done!

lionB777, 2022-11-24:
I have the same issue here as well. I just tried to install the tradfri plugin on a fresh octoprint install on a RPI4. No success too.
Thanks.

mchety, 2022-12-29:
This issue also came up for me after reinstalling octoprint. I tried Criticalstone's advice, but it did not fix the issue. The plug-in had been running fine before, but now comes up with the above errors.

ralmn, 2022-12-29:
Can you give me as much info as possible on how you reinstalled octoprint so I can try to reproduce it. Each time I can't reproduce the problem...

mchety, 2022-12-29:
I’m using a Pi 4b 4gb, which had been working fine, but I had some issues with it disconnecting from wifi randomly so I decided to grab a new sd card and flash it with octoprint via the Raspi imager (I just used the OS that was in Raspi imager under the specific purpose OSs). 

I then uploaded an octopi back up from the previous install. But when the backup loaded, the tradfri plugin was missing, so I tried reinstalling it, but got the same error messages shown above. 

I am running the Pi on a 5G wifi network, and am using a 64gb micro sd card.

Thidsa, 2023-03-04:
Hi again. Have not tried again since i tried installing many times in november last year. It worked now after updating Octoprint to 1.8.7. I dont know if that was the reason or not, but seems like it :)
Bjørn

sNaj2k, 2023-05-13:
After reinstalling my RPi4 I'm facing the same issue.

Thidsa, 2023-05-13:
I would have tried an older octoprint version and see if the plugin would work, then upgraded octoprint

sNaj2k, 2023-05-13:
> I would have tried an older octoprint version and see if the plugin would work, then upgraded octoprint

i dont think it is possible to downgrade to an older version without reinstalling everything. i upgraded to the RC of Octoprint but the issue is still there. 

Building wheel for cryptography (pyproject.toml): started
error: subprocess-exited-with-error
× Building wheel for cryptography (pyproject.toml) did not run successfully.
│ exit code: 1
╰─> [247 lines of output]

sNaj2k, 2023-05-13:
I somehow fixed it. 
While installing OctoLapse I got an Error which stated something about this Tradfri Plugin and I need to install this:

https://github.com/chrysn/aiocoap/archive/master.zip

I did and now your plugin works.

Thidsa, 2023-05-13:
Great find if that was it. Remerber Ikea plugin need something ...coap before, maybe libcoap

sNaj2k, 2023-05-13:
> Great find if that was it. Remember Ikea plugin need something ...coap before, maybe libcoap

it still cant connect to the gateway though. This whole thing is a mess.

ralmn, 2025-05-07:
I regret to inform you that I am no longer able to maintain this plugin due to time constraints and lack of access to the necessary hardware (Tradfri hub) to address issues effectively. (cf #57)

closed 2025-05-07 by ralmn