ralmn/OctoPrint-Ikea-tradfri #53: Unknown installed

HerrFrodo, 2022-12-23:
Hello together, 

whenever I try to install the plugin I get the Installation message Unknown installed. But I don't see the plugin. I tried uninstalling the plugin and deleted the folders in site-packages, but to no avail.

ralmn, 2022-12-23:
Hello,

Have you anything in the logs file ?

HerrFrodo, 2022-12-23:
I found this in octoprint.log

'''
2022-12-23 19:01:14,027 - octoprint.plugin.core - ERROR - Error loading plugin ikea_tradfri
Traceback (most recent call last):
  File "/home/pi/octoprint/lib/python3.9/site-packages/octoprint/plugin/core.py", line 1274, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/octoprint/lib/python3.9/site-packages/octoprint/plugin/core.py", line 52, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/octoprint/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/octoprint/lib/python3.9/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 711, in _load
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/octoprint/lib/python3.9/site-packages/octoprint_ikea_tradfri/__init__.py", line 12, in <module>
    import aiocoap
ModuleNotFoundError: No module named 'aiocoap'
'''

HerrFrodo, 2022-12-23:
After installing the missing package everything worked fine with the installation. Thank you.

closed 2022-12-23 by HerrFrodo