ralmn/OctoPrint-Ikea-tradfri #55: Install

wednesday-timmerday, 2024-05-23:
Hi,

when i install the "Ikea Tradfri" plugin the plugin is with success installed but when i restart the rpi the plugin does not show up in the installed plugin tab(sorry for my english)

andimae77, 2025-04-24:
Same Error here? After fresh install of octoprint via Pi-Imager, the plugin cannot be installed due to a version mismatch of aiocoap.
Prebuilt image comes with Python 3.9.2. Is there a way to force an older version of aiocoap?

Log:
INFO: pip is looking at multiple versions of aiocoap[all] to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of <Python from Requires-Python> to determine which version is compatible with other requirements. This could take a while.
INFO: pip is looking at multiple versions of octoprint-ikea-tradfri to determine which version is compatible with other requirements. This could take a while.
ERROR: Package 'aiocoap' requires a different Python: 3.9.2 not in '>=3.10'
Looks like the plugin was already installed. Forcing a reinstall.
Fehler!
Unexpected error while trying to install plugin from https://github.com/ralmn/OctoPrint-Ikea-tradfri/archive/master.zip

ralmn, 2025-05-07:
I regret to inform you that I am no longer able to maintain this plugin due to time constraints and lack of access to the necessary hardware (Tradfri hub) to address issues effectively. (cf #57)

closed 2025-05-07 by ralmn